EFeru/DbcParser #1: Multiplexing information is lost along the way

Adhara3, 2022-08-01:
Hi,

when parsing a signal the multiplexed info are lost, and this make impossible to eventually work with correct message layout.

Cheers
A

EFeru, 2022-08-01:
Hi,

Can you give a code snipped on how do you the multiplexing? 
If you look at the example in ReadMe page you should make use of bitwise OR `|=`

Adhara3, 2022-08-01:
Hi, 

thanks for the reply.
I am not doing the multiplexing myself, I am receiving data on a CAN bus, so message as `byte[]`, usually 64 bytes per payload and, given a dbc, I have to decode the values. I am using an external BitStream to read bits but I am struggling with both multiplexed signals and byte order.

The former problem, which represents my original question, is that ginev the list of signals of a message I do not have the information about which is the muxor and which channels appear given a muxor value.
I also tried using `RxSignalUnpack` but I always get 0 as eng value.

```
reader.BaseStream.BitsPosition = m_signal.StartBit;
var bits = (ulong)reader.ReadBits(m_signal.Length);
var eng = m_parser.RxSignalUnpack(bits, m_signal);
```
I'm probably doing some newbie mistake....

Thanks
A

Adhara3, 2022-08-02:
Ok, I played a bit and I may have more details

### Multimplexed
I think there is a sort of misunderstanding here. I'm not talking about how to pack several channel in a single message, but the fact thata a single message can have multiple layouts, depending on a multiplexor value.  Using your example DBC as an example

```
BO_ 1006 UI_autopilotControl: 8 GTW
 SG_ UI_autopilotControlIndex M : 0|3@1+ (1,0) [0|7] ""  APP,APS
 SG_ UI_hovEnabled m0 : 3|1@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donDisableAutoWiperDuration m0 : 4|3@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donDisableOnAutoWiperSpeed m0 : 7|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_blindspotMinSpeed m0 : 11|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_blindspotDistance m0 : 15|3@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_blindspotTTC m0 : 18|3@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donStopEndOfRampBuffer m0 : 21|3@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donDisableCutin m0 : 24|1@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donMinGoreWidthForAbortMap m0 : 25|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donAlcProgGoreAbortThres m0 : 29|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_donMinGoreWidthForAbortNotMap m0 : 33|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcDisableUltrasonicCheck m0 : 37|1@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcUltrasonicDistance m0 : 38|4@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcUltrasonicWaitTime m0 : 42|3@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcEgoLeadingReactionAccel m0 : 48|2@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcMergIntervalRearDHyst m0 : 50|2@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcMergingIntervalHeadwayHyst m0 : 52|2@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcAssertivenessRate m0 : 54|2@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_alcViewRangeSensitivity m0 : 56|2@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_camBlockLaneCheckDisable m1 : 3|1@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_camBlockLaneCheckThreshold m1 : 4|6@1+ (0.01587,0) [0|1] "%"  APP,APS
 SG_ UI_camBlockBlurDisable m1 : 10|1@1+ (1,0) [0|0] ""  APP,APS
 SG_ UI_camBlockBlurThreshold m1 : 11|6@1+ (0.01587,0) [0|1] "%"  APP,APS
```
this same message can arrive in two flavours (layouts), `m0` and `m1` depending on the muxor value (UI_autopilotControlIndex marked with M). My original question referred to this. When you parse you simply do
```
if (records[2] != ":")    // signal is multiplexed
    mux = 1;
```
but then this information is used just to offset the index for the parsing. This means that later on when message 1006 arrives I have no way to understand if it version `m0` or `m1`.
Hope this clarifies my point. Maybe I'm just missing something. 

### Helper methods to unpack
`RxSignalUnpack` requires an `uint64` as input. Usually we get data as `byte[]`. How do I convert the 8 byte `byte[]` to a properly formatted `uint64`. I tried several bit shifting (LE, BE, bit reversing...), but I couldn't find any that then gives me the correct signal value when passed as input to `RxSignalUnpack`.
```
ulong valueBitReversedLE = 0;
ulong valueBitReversedBE = 0;
ulong valueLE = 0;
ulong valueBE = 0;

var shiftUP = 0;
var shiftDOWN = 56;
for (int i = 0; i < 8; i++)
{
    valueBitReversedLE |= ((ulong)ExtensionMethods.BitReverseTable[data[i]] << shiftUP);
    valueBitReversedBE |= ((ulong)ExtensionMethods.BitReverseTable[data[i]] << shiftDOWN);
    valueLE |= ((ulong)data[i] << shiftUP);
    valueBE |= ((ulong)data[i] << shiftDOWN);
    
    shiftUP += 8;
    shiftDOWN -= 8;
}

Thanks a lot
A

```

Adhara3, 2022-08-02:
Helper method to unpack, works, sorry for the noise.
The rigth way is this one.
`valueLE |= ((ulong)data[i] << shiftUP);`

The question about muxed is still valid but next week I will be on vacation, I can provide a pull request

Cheers
A

EFeru, 2022-08-05:
Hi @Adhara3 ,
Sorry for the late reply.
- I understand now what you meant by muxing. As you you also found out muxing is not supported at the moment. Feel free to create a pull request and I can support you.
- To convert `byte[] myData` to `ulong` you can also use the `System.BitConverter` class in C# and then call:
```
ulong myDataULong = BitConverter.ToUInt64(myData, 0);
```

Adhara3, 2022-08-07:
Hi @EFeru ,

in the next couple of days I will have some spare time and I will spend some on this project. Here is my plan:

### Test project
Add a test project to the solution where I can put at least some very basic non regression tests to start with., then add tests while I add new features 

### Split classes
I would like to split the current `DbcParser` class in at least 3 classes

1. A static `DbcParser` class with only methods for parsing
2. A static `DbcPacker` class with methods to pack/unpack signals for sending
3. A non static `Dbc` class, which is the result of a parsing

The reasons for this is first of all to split responsibilities and will also make all classes immutable.
The obvious cons here is that it breaks current API, unless we call the parsing something different like `DbcParserUtility` and keep the original `DbcParser` class but marking it as `Obsolete` and in it we call the new methods so that the logic is in one place.

### Features, fixes
I think (need to verify writing a  UT) there is a bug on comments, where you search the whole file for a signal name ignoring the message ID. This may be broken when different messages carry the same signal name. Moreover, this way you must read the full file upfront which should not be necessary at all with a DBC file.

Having done this, I would proceed with a line by line parsing, so that we can avoid the `.ReadToEnd()` call on the stream, favoring a `.ReadLine()` (so we could also get rid of the custom find new line part in the code) approach not only for performance reasons (quite limited benefits though, for usual dbc files which are not huge in size, but nevertheless a good practice) but also to allow real time parsing (network stream)
Next, add a `.ParseFrom(Stream stream)` method, I think it may be useful, streams are very generic and can go from memory stuff (a not persisted yet file maybe), to network stream, to compressed streams (think about reading a dbc from a zip file).

Little foreword before going on here.
I noticed that you kept the properties in that class 1:1 with the file, e.g. endianness is an integer with 0 and 1, the meaning of 0 and 1 must be known by the user, you didn't use a `public bool MSB {get;}` property to the class. Or value labels are left as they are, a string to be parsed and splitted by the user. I think there may be advantages in keeping properties so raw (they really indicate file content with no extra magic) but I would then add some extension methods to both `Signal` and `Message` classes to quickly get more user friendly properties. I would not add them into the class itself to avoid cluttering. 
Some examples:
`Msb`, `Lsb`, `IsInteger`, `ValueLabelsAsDictionary`, `MultiplexingInfo`, `HasMultiplexing` (on `Message` class)

Add support for multiplexing. At least simple, not nested multiplexing. Following what I just mentioned, `Signal` would have a new `public string Multiplexing {get;}` property, that will be filled with `M`, `m1` etc.. i.e. the real content of the file and then add the aforementioned extension methods to retrieve a more complete and parsed multiplexing info.

Last but not least, add support for floating point numbers. There is a standard way in the dbc to tell that a signal 32/64 bit should be interpreted as float/double. This will also imply a change in the `DbcPacker` class I guess and here I may need your help.

What do you think about this? Should I proceed? Are you interested in having this in your library? Thoughts?
Thanks
A

EFeru, 2022-08-26:
Hi @Adhara3,
I think you have very good ideas from improvements. I want to catch up in the next days. Feel free to continue. Are you going to implement the above in #2 ? So, I can follow it up.

Adhara3, 2022-08-28:
Hi @EFeru 

> Are you going to implement the above in #2 ? So, I can follow it up.
Everything indicated above is available in the PR, except the `SIG_VALTYPE_` thing, for that I opened a separated  issue #3
Moreover I just miss a couple of tests I will push today.

Cheers
A

Adhara3, 2022-08-30:
I think this issue can be closed, the muxing info is now in and the refactoring is ok

closed 2022-08-30 by Adhara3