EFeru/DbcParser #3: Add support for float and double value type

Adhara3, 2022-08-24:
Hi,

as already mentioned, I would like to add this feature based on the `SIG_VALTYPE_` tag.
I will take care of adding it once we are done with the refactoring.

I just wanted to share a related discussion here. `Signal.IsSigned` as of today is of type `byte`. This is strange, given that usually a property name starting with 'is' usually identifies a `bool`. However, adding floating point values make the `IsSigned` property obsolete, since it has no meaning for floating points.

My suggestion is to remove the `IsSigned` property and replace it by a `ValueType` property, defined as follows:
```cs
public class Signal
{
    // ...
    public DbcValueType ValueType { get; }
   // ...
}

public enum DbcValueType
{
    Signed, Unsigned, IEEEFloat, IEEEDouble
}
```

this follows what happens in Vector CANdb++ editor which I usually take as a reference given the fact Vector is DBC "father".

What do you think about it?

Cheers
A

EFeru, 2022-08-26:
We can follow the CANdb++ approach, I agree. Sounds like a good idea.

EFeru, 2022-10-28:
Hi @Adhara3 , are you willinig to support on this?

Adhara3, 2023-01-09:
> Hi @Adhara3 , are you willinig to support on this?

I'll work on this issue this week

EFeru, 2023-01-14:
@Adhara3 what do you say if I make you a collaborator for this repo?

Adhara3, 2023-01-14:
> @Adhara3 what do you say if I make you a collaborator for this repo?

It would be a privilege.
I currently need this library for some work activities so in the short/mid term I will provide feedback and effort on it.
I do not expect to last forever, also because DBC spec are quite stable, so I think we will soon be quite complete.

I found this document that is quite detailed and I think covers many parsing details and contains the answer for #15 (see chapter 12) but also #21 (see chapter 9).

Cheers
A

Adhara3, 2023-01-17:
@EFeru on the branch for this issue I put the line parser and stuff. We need to implement the packer and unpacker part.
Do you know how to do it?

A

EFeru, 2023-01-18:
Hi @Adhara3,  The packer and unpacker already work with input/output type as `double`.
https://github.com/EFeru/DbcParser/blob/21ee82798f00795aa926d03b0040dcf241f26879/DbcParserLib/Packer.cs#L22

But maybe I did not fully understand your question.

taylorjdlee, 2023-01-18:
What good timing I'm facing an issue with signed signals while trying to unpack messages using the packer class.
Here's an extract of the .dbc I'm using: https://drive.google.com/file/d/1RJc9zE_VEt6jdlhrABVl-E-nyLOPOi5l/view?usp=share_link

The main things to keep note of are the Latitude, Longitude and Altitude signals (which are all signed). No matter what I feed to the packer class the value returned is 0 for all 3 signals. 

Maybe this is because the factor used for them is in scientific notation instead of a decimal? i.e for Latitude the Factor is 1e-16

@Adhara3 Hopefully the .dbc file extract helps you in solving this issue

Adhara3, 2023-01-18:
> Hi @Adhara3, The packer and unpacker already work with input/output type as `double`.
> 
> https://github.com/EFeru/DbcParser/blob/21ee82798f00795aa926d03b0040dcf241f26879/DbcParserLib/Packer.cs#L22
> 
> But maybe I did not fully understand your question.

The problem is that RxSignalUnpack does not take into account if the underlying bits already represent a floating point. 
It just checks for IsSigned but it is not enough. If the 32 bit are a single you cannot simply cast to double, you need a sort of reinterpret cast. I'll try to push something on the branch to show you

EFeru, 2023-01-18:
@taylorjdlee can you also give the code piece where you unpack the messages with the issue?

closed 2023-02-07 by Adhara3