EFeru/DbcParser #4: Evaluate using Regex for parsing lines

Adhara3, 2022-08-24:
Hi,

I am not a huge regex fan, but I have to admit that this line parsing really looks like a regex playground.
Some reasons, not in priority order:
-  `string.Split` does allocate arrays. Not a big deal, but...
-  `.Substring`, `.Join` `.Trim` all create other strings, often just temporary
-  Regex allows a more strict syntax check
-  Regex allows a safer handling of spaces (`\s` works for both `" "` and  `"\t"`)

There are scenarios where the `string.Split` is a bit tricky, e.g. 
```
SG_ DAS_steeringControlType : 23|2@0+ (1,0) [0|0] ""  EPAS,TEST
```
supports multiple receivers, BUT in Vector CANdb++ this is also supported
```
SG_ DAS_steeringControlType : 23|2@0+ (1,0) [0|0] ""  EPAS, TEST
```
with a spacing after the comma. This breaks current implementation.

I will try to add a benchmark project in the solution just to make sure performances do not dramatically drop. 
I will provide a separate PR for this item so that you can evaluate other aspects like readability.
Moreover, for this kind of refactorings having unit tests is a must.

Cheers 
A

EFeru, 2022-10-28:
Hi @Adhara3 , Is this already implemented ?

Adhara3, 2023-01-15:
@EFeru I did for some simple line parsers, others are a bit more complicated, we can migrate later on.
Anyway following [this document](http://mcu.so/Microcontroller/Automotive/DBC_File_Format_Documentation.pdf) it would be easier to migrate. There are some interesting constrains that should help (like names can only start with letter or underscore and cannot be longer than 128 chars). 

Anyway, as a general rule, I wouldn't give this a high priority for two reasons

1. It does not involve public API, it is just implementation so it could be changed at any time with no harm. And being the _behind the scenes_ part, it should not impact future developments in any way
2. We are covered with tests, so any change in how we parse should be quite easy and safe

Cheers
A

closed 2023-03-24 by Adhara3