EFeru/DbcParser #6: `Parser` class and methoda are static

EFeru, 2022-08-31:
Hi @Adhara3 ,

Is there a reason why we made the `Parser` class and methods static:
https://github.com/EFeru/DbcParser/blob/a47d31f80b52a1b68273ca65fdd10a30726a6cbe/DbcParserLib/Parser.cs#L7-L9

This will prevent for doing somethig like this:
```cs
Parser dbc = new Parser();  // user cannot do `new` on static classes
dbc.ParseFromPath(filePath);
```
instead the user needs to do:
```cs
Dbc dbc = Parser.ParseFromPath(filePath);
```

I am not sure if it has any impact on flexibility at the user side. Let me know what is your opinion.

Adhara3, 2022-09-01:
Let's put t in another way. Why would you instantiate an object with no state? Which is the benefit?
`Parser` and `Packer` are static because they are helper methods that are part of a class just to make them more readable. Moreover, static methods in this case are even more efficient without requiring any garbage collection.

Having said that, feel free to change the class and make it non static, if you prefer. But there is no technical advantage in doing so. No real harm either.

There would be a difference, as an example, if you wanted to configure your parser i.e. if you need to pass in something in the constructor and the parser has then a state. In that case you would require to do what you are saying.
```cs
var parser = new Parser(configuration);
```

Hope it helps
A

EFeru, 2022-09-01:
For me is no problem to keep it this way. I just wanted to check with you if there are any drawbacks. From what I hear we should be fine. So I'll close the issue. Thanks!

closed 2022-09-01 by EFeru

Adhara3, 2022-09-01:
There is an interesting point here. After the refactoring the parsing process **could** be extended by the user.
At the moment we are not 100% DBC spec compliant, there are some "tags" we are ignoring.

Now, imagine John, a lib user, is parsing a DBC but requires a specific line to be parsed but we are ignoring it. He is stuck. He must open an issue or submit a PR and wait for a new release.
But it does not have to be like that. 
This is happening because `Parser` is static and as such is internally using a list of line parsers that is also static and private.
If John had access to that list, he could:
- write his own `ILinePaeser` implementation that parses the line he needs
- add this parser in the list of parsers
- let the parser use the new list.

He is basically _extending_ the library himself.
Hope this is clear.

How to achieve this?
Today the fact that `Parser` is using a list of `ILineParser` to fulfill his job is an _implementation detail_ it is hidden, nobody knows outside the lib. This has a clear advantage which is: we can internally change it any time without breaking the API. To let the user extend the lib, this can no longer be an _implementatuon detail_.
Moreover, in this scenario we could:
- keep `Paeser` static, make the list public so that the user can set it. It works, not thread safe
- make `Paeser` no longer static
```cs
// Parameterless constructor is using default list
var standardParser = new Parser();
var myLineParsers = new List<ILineParser>
{
    // My parsers
};
// Or explicit list.
var customParser = new Parser(myLineParsers);

// If we keep it static
Parser.LineParsers = myLineParsers;
var dbc = Parser.Parse(...);

// The non static approach would allow 
// Independent parallel parsing 
// With different set of parser. 
// Cool but ..  real world scenario?
```
See? That would be powerful and I hope this make clear why it is static now.
If you'd like to make the parsing extendible by the user, maybe open an _ad hoc_ issue.
A

PS: the user extending is possible only because of the Object Oriented refactoring