EFeru/DbcParser #9: Add netcoreapp3.1 support

taylorjdlee, 2022-09-22:
While trying to address issue #7 I decided to try building the library with netcoreapp3.1 and it does build successfully. This is as the project I'm using the library on uses the netcoreapp3.1 framework

Should be a simple addition to the framework section of the DbcParserLib.csproj

hakwes, 2022-09-22:
@taylorjdlee According to [Microsoft docs](https://learn.microsoft.com/en-us/dotnet/standard/net-standard?tabs=net-standard-1-0#:~:text=However%2C%20if%20you%20want%20to%20share%20code%20between%20.NET%20Framework%20and%20any%20other%20.NET%20implementation%2C%20such%20as%20.NET%20Core%2C%20your%20library%20should%20target%20.NET%20Standard%202.0) the project should target netstandard2.0

EFeru, 2022-09-30:
Fixed see #11

closed 2022-09-30 by EFeru