EFeru/DbcParser #14: Improved lookup

hakwes, 2022-09-29:
I would like to see the Messages property of the DBC class to be set as a Dictionary to improve lookup. 
Another option would be to add a Get method that would use an internal dictionary.
Another suggestions is to att the ValueTable as a dictionary as a complement to the string. That would simplify mapping values to descriptions.

EFeru, 2022-09-30:
Sounds interesting. I din't use Dictionary in C# yet, so if you are willing to help I would be more than happy

taylorjdlee, 2022-09-30:
I guess this also somewhat ties into issue #15

hakwes, 2022-10-03:
@EFeru I've made a workaround in a[ fork,](https://github.com/hakwes/DbcParser/blob/currentinternalrelease/DbcParserLib/DbcBuilder.cs#:~:text=%7D-,public%20void%20LinkTableValuesToSignal(uint%20messageId%2C%20string%20signalName%2C%20string,%7D,-private%20bool%20CheckExtID) 

But I that logic should be put in the ValueTableLineParser so the line isn´t parsed twice.

EFeru, 2022-10-28:
@hakwes Are you willing to support on converting to Dictionary? Especially the Value Tables would be really nice to have them as Dictionary.
I can help you updating the Tests when you have the PR ready.
For now I did a quick fix on Value table because it could not parse properly the spaces see #13 .

hakwes, 2022-11-01:
Sure, I'll have a go at it @EFeru

Adhara3, 2023-01-14:
Hi,

personally I would not change Node and Message container type. 
The c# language provides easy ways to make a list a dictionary using what the user likes as a key.
Moreover, forcing a unique key would not work with the composite dbc as in #20 because multiple files may contain the same ID. 
Finally, a dbc file may contain the same ID twice. This would brake the whole parsing. Would this be ok? I don't know.

If we wish to _extend_ `Message` providing non standard properties as key value pair sounds ok to me. 
Can we start by having a string-string map?

I think the goal of this library is to parse a text file and put stuff into usable object. Keep it simple. 
If you need extra features do not force them into the parsing process, instead let's add validators or helpers around the parsed objects.

My 2 cents
A

Adhara3, 2023-01-14:
> I guess this also somewhat ties into issue #15 making message a dictionary would help in solving this issue as well.

I fear the two things are unrelated

A

EFeru, 2023-01-14:
Hi @Adhara3 , thanks for your detailed answer. Is very good to have both advantages and disadvantages in our overview.

Adhara3, 2023-01-14:
> 

So just to recap:

- dictionary vs try get, still under investigation. For proper API the same should apply to Messages (by Id)  Nodes (by name) and Signals (by name) in a message especially if we want to reflect the spec
- for `ValueTable`, yes. It is a string because we left the _pure parsed value_ but should become a dictionary or similar.

Cheers
A

hakwes, 2023-01-25:
@Adhara3 I don't really understand your concerns about exposing the Message property as a Dictionary. 
Internally, messages are added to a Dictionary first and then the Dictionary is converted to an Array.
A Dbc file can be very, very large and adding an index method where you would parse the file again seems excessive.

The TryGet method is fine but I don't think you need to indexDbc class to implement that.

Uight, 2024-07-29:
@EFeru, @Adhara3 
I would have a question on this. What are your usecases for this nuget package. For me it is reading in the dbc file once and then using it to interpret incoming can messages. We can have 500-10000 Can messages per second and for every receive message you must find the message and then read the signals. The only useable way to make this lookup fast enough is to have a dictionary. 
As this is a parser and not an editor i think having a fixed (readonly) dictionary could be a good way. I would therefor suggest making messages a IReadOnlyDictionary. 

but i would also be interested in other usecases to see if there might be a better solution.

Uight, 2024-07-29:
I was just going through the code to check out how using a IReadonlydictionary for messages would change the code. (pretty much only the tests)
In the DbcBuild.cs i found this comment: //TODO: uncomment once Immutable classes are used
Is it the plan to make everything in the Dbc class immutable? If yes i would really suggest using a IReadOnlyDictionary for the Messages and i could do some of the immutable stuff to when im at it. I just think its impossible to make everything immutable in one swoop but it could atleast start it for messages and nodes. I would use a IReadOnlyCollection on the top level for nodes then. Whats your thoughts on that?

Uight, 2024-08-01:
@Adhara3 @EFeru,
 
i know changed to code in a way that the dbc Class is the same as before but you can now convert to a immutable dbc class that works with the immutabel objects and dictionaries. The immutable classes the support some more functions as they cant be changed, which solves #37 and is a part of solving #68.
I changed the dbc builder test to test the immutable classes while i kept the parser test working on the old dbc class.
this could be a way to keep open for the future. If you want to edit the dbcs in code you would use the old classes and if you dont and want fast access you just use the immutable classes.
However im not sure if this is really needed as i have a feeling that this is only done because of fear of a breaking change. And this introduces some other problems with classes like the Packer which currently work on signals while i would want to work them on immutableSignals. You could just use another wrap there to make both possible but it just increases the work when changing stuff.

Adhara3, 2024-08-26:
> @EFeru, @Adhara3 I would have a question on this. What are your usecases for this nuget package. For me it is reading in the dbc file once and then using it to interpret incoming can messages. 
I think someone in the past asket to _add messages_ but the usage requested was not clear since no `.save()` method was requested.

At the moment the lib has been designed primarily with this exact scenario in mind. 
Our design concern has been to avoid having objects that can do everything. The DBC class should only be an in memory "browsable" representation of a dbc file with probably no optimizations in it. The reason is that if it is optimized for read, then it would not work well for write (e.g. IReadOnly is fine for a non editable object only). 

The Packer part has been kept separated because if is another kind of functionality, not directly related with the _dbc in memory representation_ .  So we could think of a method of some kind that takes the generic Dbc object and `.ToOptimizedRead()` to get a new optimized version of the DBC for reading and that could even embed the packer part.

The overall design can be discussed, for now the lib is designed to work best on the scenario you described, I would not close the door to future developments to support other scenarios this is why I would not force stuff on the DBC class itself.

Hope this clarifies the intent

Cheers
A

EFeru, 2024-08-26:
Hi @Adhara3,
My use case is the same as @Uight parse the dbc once and then unpack/pack incoming/outgoing mesages. Is your use case different?
If not, then I would also go for an implementation that maximizes the throughput. So, if Dictionaries is the way to go for speed in getting the Message by id, I would sugest we do that, or we provide helpers that do that. In essence we need something to get the messages by ID quickly, :
```cs
dbc.Messages.TryGetValue(myID, out Messsage myMessage)
```
I think `Dictionay` complexity `O(1)`, is faster than `IEnumerable` complexity `O(n)`:
```cs
Messsage myMessage = dbc.Messages.Where(m => m.ID == myID);
```

Regarding loading multiple dbc files and clash between IDs, please correct if I am wrong, I think they should be treated as different dbc objects `dbc1`, `dbc2`, ... `dbcN`.

Uight, 2024-08-26:
I think the change to dictionary is kind of obvios here as the dictionary is also used internally. the way i did is that internally a dictionary is used and externaly this dictionary is readonly. 
this shouldnt realy affect speed in both cases. It only means that the dbc is more or less immutable from extern. But you could always add dedicated methods for changing objects which would in my opinion be teh better way anyway as it allows for validation in the methods and not just resetting/overwriting properties. 
I would also argue that a normal dictionary will be good enough speed wise in 95% of the cases. We are talking of c# anyway ;)
One thing could be the frozen stuff added to .net8 but that wouldnt work in .net framework.

The thing i changed two was the properties because you could easily get null ref exceptions there because only one of the properties was set at any given time (either string, enum, double....). I think this change could be discussed