EFeru/DbcParser #21: Can （ Environment variables）be supported for parsing dbc files?

mtooooo, 2023-01-05:
As the title

EFeru, 2023-01-10:
Hi @mtooooo, Sorry, I don't fully understand. Can you be more specific on the needs you have?

Adhara3, 2023-01-14:
@EFeru dbc can specify environment variables using tag `EV_`
We can have a look at this. Lower priority, but can be done.

A

mtooooo, 2023-01-28:
Vector CANdb++ Editor show.
[Snipaste_2023-01-28_17-16-21.jpg](https://p1-juejin.byteimg.com/tos-cn-i-k3u1fbpfcp/377f97d98b2842d3870bca5f57da7773~tplv-k3u1fbpfcp-watermark.image?)

Whitehouse112, 2023-02-15:
@mtooooo can you provide the entire .dbc file shown in the image?

I tried to copy your `EV_` lines into my test file but i get a .dbc file that can't be opened with Vector CANdb++ (version 3.1.21) Which version of CANdb++ are you using?

mtooooo, 2023-02-16:
Dear all,

     

        I'm use  CANdb++ Version 3.1.21.     EH6466.dbc file is attached to the mailbox。

Thanks!

Adhara3, 2023-02-16:
Please @mtooooo , do drag and drop the dbc file here, in your reply message. 
The file should get attached to your message.

Thanks

mtooooo, 2023-02-16:
Re: [EFeru/DbcParser] Add support for Environment variables when parsing dbc files (Issue #21)
-----------------------------------

 

 
Please @mtooooo , do drag and drop the dbc file here, in your reply message.
 The file should get attached to your message.
 
Thanks

mtooooo, 2023-02-17:
[EH6466.zip](https://github.com/EFeru/DbcParser/files/10762715/EH6466.zip)

mtooooo, 2023-02-17:
using Vector CANdb++ (version 3.1.21).

closed 2023-03-24 by mtooooo