EFeru/DbcParser #22: Signal ValueTable should not be a plain string but a more usable object instead

Adhara3, 2023-01-15:
As requested in issue #14 by @hakwes, I create a separate issue here because the impact is different and this can be completed faster

EFeru, 2023-01-15:
I fully agree with you that plain string is not very useful. A Dictionary might fit nicely for this purpose. What do you think?

Adhara3, 2023-01-16:
@EFeru I already mentioned all this in a part in [this](https://github.com/EFeru/DbcParser/issues/1#issuecomment-1207354189)  post back in August. I quote here:

> I noticed that you kept the properties in that class 1:1 with the file, e.g. endianness is an integer with 0 and 1, the meaning of 0 and 1 must be known by the user, you didn't use a `public bool MSB {get;}` property to the class. Or value labels are left as they are, a string to be parsed and splitted by the user. I think there may be advantages in keeping properties so raw (they really indicate file content with no extra magic) but I would then add some extension methods to both `Signal` and `Message` classes to quickly get more user friendly properties. I would not add them into the class itself to avoid cluttering. 
Some examples:
`Msb`, `Lsb`, `IsInteger`, `ValueLabelsAsDictionary`, `MultiplexingInfo`, `HasMultiplexing` (on `Message` class)

So the fact that some properties are left as they are was a choice. Is still valid? If as a design choice you want to keep properties raw, then we should leave labels as string and provide an external method to optimize. Otherwise, we should migrate every property to a more usable one (e.g. endianess should not be int but an `enum` instead)

BTW, for labels an extension method already exists that splits the string and provides a convenient way to iterate over splitted values `.ToPairs()`
```cs
public static IEnumerable<KeyValuePair<int, string>> ToPairs(this Signal signal)
{
  if (string.IsNullOrWhiteSpace(signal.ValueTable))
	yield break;
	
  using(var reader = new StringReader(signal.ValueTable))
  {
	while(reader.Peek() > -1)
	{
	  var tokens = reader.ReadLine().Split(' ');
	  yield return new KeyValuePair<int, string>(int.Parse(tokens[0]), tokens[1]);
	}
  }
}
```
I know it-s not a dictionary, I-m more interested here in the design choice. Property or extension method?

Let me know, thanks
A

Adhara3, 2023-01-16:
Let me add i note which is related to #23 .
As it is today we do not validate the table value, as an example
`VAL_ 2297441534 MaterialDropActiveStatus 3 "NotAvailable" 2 "Error" 1 "On" 0 "Off" ;`

This is good, but what happens if the line is
`VAL_ 2297441534 MaterialDropActiveStatus 3 "NotAvailable" 2 "Error" 1 "On" 1 "Off" ;`
The 1 value is defined twice.

Now two things happen if you want to have a dictionary up front

1. The problem should be spotted and signalled to the user as a parsing problem (because the line is invalid)
2. Today the `.ToPairs()` method works because it does not force a unique key concept (I did that on purpose), but with a dictionary then the key is unique and you then need to choose which one to keep. Note that any choice here is inherently wrong and right at the same time  as long as it is documented and the parsing issue is notified to the user.

Just to point out that _using dictionaries_ is not just a container choice, it has a broader impact on what has to be done during parsing.

Cheers
A

Adhara3, 2023-01-16:
Please note that multiplexing is following the same pattern.
`signal.Multiplexing` is a `string`,  the more usable multiplexing info is built using the extension method `signal.MultiplexingInfo()`

Here again, the property is raw, the access to usable data is external.

A

EFeru, 2023-01-21:
I am in for Value pair approach. Not sure if Vector db or other tools allows duplicates, but like this we are on the safe side.

Adhara3, 2023-01-23:
> I am in for Value pair approach. Not sure if Vector db or other tools allows duplicates, but like this we are on the safe side.

Wait, it's easy to be on the safe side, my point was just that switching to a strict key-value (with unique key) container implies some work on the validation part. Once we do that, we are safe.

> Not sure if Vector db or other tools allows duplicates
I just checked and VECTOR's CANdb++ does allow duplicates (which is weird nonetheless).

Cheers
A

hakwes, 2023-01-25:
I made a workaround for this in a fork.
https://github.com/hakwes/DbcParser/blob/currentinternalrelease/DbcParserLib/DbcBuilder.cs#:~:text=private%20Dictionary%3Cdouble%2C%20string%3E%20CreateDescription(string%20valueTableLine)
But it would be more efficient storing the values in a dictionary from the start.

I checked the Kvaser Database Editor and that does not allow duplicates. I would expect that to be the correct behaviour.

closed 2023-02-14 by Adhara3