EFeru/DbcParser #23: Parsing failures management

Adhara3, 2023-01-16:
Today we do silently swallow errors when parsing.
My idea is to provide an optional `IParsingObserver` which will allow user to react (or at least know!) to errors.

Quick example
```cs
public interface IParseObserver
{
  void DuplicateMessage(int messageId);
  void DuplicateSignalInMessage(int messageId, string signalName);
  void DuplicateValueTableKey(double key, string value);
  // etc...
}

public class StrictParseObserver : IParseObserver
{
  public void DuplicateMessage(int message)
  {
	// This implementation is just an example
	// The idea is that the parsing here will stop and fail
	throw new ParsingException($"More than one message in dbc is defined with ID {message}, which does not follow the official spec");
  }
  
  // Etc...
}

// We could also provide an implementation that does nothing, like today
public class SilentParseObserver : IParseObserver
{
  public void DuplicateMessage(int message)
  {
  }
  
  // Etc...
}

// But a user could inject something custom like this
public class StrictParseObserver : IParseObserver
{
  public void DuplicateMessage(int message)
  {
	// I don't want to break, I just want to log the issue to create a report
	Console.WriteLine($"More than one message in dbc is defined with ID {message}, which does not follow the official spec");
  }
  
  // Etc...
}
```

Hope you get the idea. 
A

EFeru, 2023-01-21:
Interesting proposal. Looks like a way to provide consistency checks. I would say if there is no issue on increasing too much the parsing time, i mean why not.

Adhara3, 2023-01-23:
> Interesting proposal. Looks like a way to provide consistency checks. I would say if there is no issue on increasing too much the parsing time, i mean why not.

A successful parsing will pay no extra time

A

closed 2023-09-27 by Adhara3