EFeru/DbcParser #32: Enable Regex Parsing?

JDDEmbedded, 2023-04-13:
How do I enable the new regex parsing feature? Do I have to use the source code or is there a way to do it with v1.3.0 as compiled? Right now the SignalLineParser, at least, is not using the regex parsing by default. I've found that some DBC files will not parse properly. For example, if there is no space between the signal name and the ":" character then an exception is thrown. A quick test with an online regex tester shows that this issue won't happen with the regex parsing. Would be nice if Parser.ParseFromPath or similar methods had a parameter that allowed the caller to specify whether to parse using regex.

Adhara3, 2023-04-14:
Hi,

regex parsing is enabled by default in verssion 1.3.0, if it is failing there is a bug in the regex we are using.
Please provide an example of failing parsing, so we can tesst and fix.

Thanks
A

Adhara3, 2023-04-14:
Hi,

we double checked, you are right `SsignalLineParser` is still using the legacy parsing, not the regex based one. Thiss is a leftover, we will release ASAP a fixed version.

> Would be nice if Parser.ParseFromPath or similar methods had a parameter that allowed the caller to specify whether to parse using regex.

We won't do this, we prefer to keep it simple and use regex by default everywhere as the only parsing strategy. 
In the future, if we ever introduce a new different parsing strategy, we will allow user to select which one to use.

Sorry for the inconvenience
A

Adhara3, 2023-04-14:
Hi, 

released in version 1.3.1 alreay available on [nuget.org](https://www.nuget.org/packages/DbcParserLib/1.3.1)

Cheers
A

closed 2023-04-14 by Adhara3

JDDEmbedded, 2023-04-14:
Thanks for the prompt fix!