EFeru/DbcParser #34: Introduce "SendType" for Message and Signal.

Usamaahmad751, 2023-04-29:
While parsing a DBC file at my work using this parser library I realized that this library doesn't support parsing of "SendType" for both Signal and Message. For our particular project we are using this information from the DBC file. So, I modified this library a bit and added a "SendType Parser" that gives the assigned "SendType" information for each Message and Signal otherwise it will return default value as "Cyclic".

**The available send types are defined in the DBC at:**

`BA_DEF_ BO_  "GenMsgSendType" ENUM  "Cyclic","Event","NotUsed","NotUsed","NotUsed","NotUsed","NotUsed","IfActive","NoMsgSendType","NotUsed","NotUsed","NotUsed","vector_leerstring";
`

**These can change from DBC to DBC for example the following is also possible:**

`"cyclic","triggered","cyclicIfActive","cyclicAndTriggered","cyclicIfActiveAndTriggered","none" `

**The Message "Send Type" can be found here:**

`BA_ "GenMsgSendType" BO_ 1424 7;`

**Where "BO_ 1424" is the id of the message and "7" is the Enum value** 

**If a message has no definition of BA_ "GenMsgSendType" BO_ XXXX X  we should use the default value which is defined here:**

`BA_DEF_DEF_  "GenMsgSendType" "Cyclic";`

**Also each signal can also have a "SendType" Information that can be found here:**

`BA_DEF_ SG_  "GenSigSendType" ENUM  "Cyclic","OnWrite","OnWriteWithRepetition","OnChange","OnChangeWithRepetition","IfActive","IfActiveWithRepetition","NoSigSendType","OnChangeAndIfActive","OnChangeAndIfActiveWithRepetition","NotUsed","NotUsed","NotUsed";`

`BA_ "GenSigSendType" SG_ 1424 SeatControlCommand1 8;`

**SeatControlCommand1 is the Signal Name and "8" is the ENUM value.**

**If this entry is not present the default value shall be used:**

`BA_DEF_DEF_  "GenSigSendType" "Cyclic";`

Adhara3, 2023-05-15:
Hi, 

did you have a look into the `CustomProperites` map available on both `Signal` and `Message` class?
Can you confirm that you can find it there?

Thanks
A

Adhara3, 2023-10-02:
@Usamaahmad751 any news on this? Did `CustomProperties` answered your questions?
Please let us know, or give some more feedback, otherwise I will close this issue,

Cheers
A