EFeru/DbcParser #37: IsMultiplexed run time

taylorjdlee, 2023-07-17:
https://github.com/EFeru/DbcParser/blob/59b4eb633e4fdf7f02858b2e27484be89287a570/DbcParserLib/Extensions.cs#L64

Due to using the any() method IsMultiplexed() can take sometime to run if there are a lot of signals present in the message. It would be great if we could have a simple boolean in the message that means we don't have to iterate across every signal

Uight, 2024-07-26:
I pretty much agree with @Adhara3 on this but i guess it could be helpfull to have a IsMultiplexed Property instead of the ExtensionMethod. I wouldnt be using a method for something that cant change. 
Its pretty much just a nice thing as then you wouldnt have to store it in your own "wrapping" class for message. 

Otherwise theres not much to optimize the code of IsMultiplexed(). However if you are using Sonarqube it suggests using Exists instead of Any. (https://sonarsource.github.io/rspec/#/rspec/S6605/csharp)
But i doubt that would make a difference as the Signal list is always pretty small. Havents seen any can message with more than 30 signals.

Adhara3, 2024-08-31:
Will be part of the immutability refactoring and will I Claude the extended multiplexing