EFeru/DbcParser #41: EnumCustomProperty parsing fails if more than one white-space character is used as separator

JuRichter, 2023-08-14:
Line in dbc file:
`BA_DEF_ BO_  "GenMsgSendType" ENUM  "Cyclic","Event","CyclicIfActive","SpontanWithDelay","CyclicAndSpontan";`

Propose fix in PropertiesDefinitionLineParser.cs:
Line 16:
Add:
`private readonly char[] PropertySeparatorChar = { ' ' };`

Line 92:
Replace:
`var enumDefinition = match.Groups[10].Value.Replace("\"", "").Split(' ')[1];`
By:
`var enumDefinition = match.Groups[10].Value.Replace("\"", "").Split(PropertySeparatorChar, StringSplitOptions.RemoveEmptyEntries)[1];`

Whitehouse112, 2023-08-25:
Hi JuRichter,
thanks for the feedback. The regex to parse the Enum custom property has been modified to correctly read more white-spaces without the need of the Split() function. This will be fixed in the next release.

Regards

JuRichter, 2023-08-25:
Thanks a lot!

Brainy0207, 2023-09-20:
One of the last changes to the regex for parsing enum definitions caused a strange issue.

Assume this definition:
`BA_DEF_ BU_ "AttributeName" ENUM "Val1","Val2","Val3";`

Parsing this definition would duplicate the last value.
`.Values = { "Val1", "Val2", "Val3Val3" };`

I submitted a PR (#46) to fix the issue and add some test code.