EFeru/DbcParser #43: Default CycleTime(BA_DEF_DEF GenMsgCycleTime) should be set as CycleTime before individual CycleTime(if existing) set as CycleTime

SebastianBolte, 2023-08-25:
"BA_DEF_DEF_: Attribute default definition. Specifies the default value for the attribute if none is given."

This means that the CycleTime should be initialized with the DefaultCycleTime if there is no definition for the CycleTime of a specific message

After parsing a dbc, the user gets a list of messages. Each message contains i.a. the CycleTime, although there is no information as to whether the value of the CycleTime was set by the class initialization or by an entry in the dbc. The DefaultCycleTime is available as a CustomProperty. Information about the decision criterion as to whether the CycleTime must be overwritten with the DefaultCycleTime "...Specifies the default value for the attribute if none is given." is not present. If you do not allow the value 0 as a valid CycleTime, you can check the CycleTime for the value 0 to get this information, since the CycleTime is set to the value 0 via the class initialization. If it contains any other value, there must have been an entry for the CycleTime of that message in the dbc. But I would like to allow the value 0 as a valid CycleTime. Perhaps the parser could first initialize each message with the DefaultCycleTime. If there is still an entry for a specific CycleTime in the DBC, the CycleTime can be overwritten with it. This would of course be easier for the user of the DbcParserLib, but the question arises as to how much logic the parser should contain. Otherwise it would be great if the Message class would contain a flag indicating whether there was an entry for the CycleTime. Thanks in advance.

Best regards,
Sebastian

Adhara3, 2023-09-01:
Hi,

the property is there because it was historically available _even before custom attributes were supported_. Basically, this is a special case.
My suggestion is to ignore the property, instead look into the CustomProperties. If `CycleTime` has been set, then the property is available, otherwise the property does not exist.

That should be at least a working solution, moreover I would like to completely remove the property being, as said, a special case. Maybe I could expose it as extension method.

Hope this helps
A

Whitehouse112, 2023-10-20:
Hi @SebastianBolte,
from this release the _CycleTime_ property is no longer a special case but it's treated as a "normal" custom property. You can easily get its value with `bool CycleTime(this Message msg, out int cycleTime)` to know whether this property was set from dbc or not, and if so, get its value. The _CycleTime_ Message property is now marked as obsolete, we plan to removed it completely in future releases.

Regards

closed 2023-11-06 by Adhara3