EFeru/DbcParser #65: Ignore Minumum Maximum values of CustomAttributes when both are set to 0

JuRichter, 2024-06-24:
Hi, I'm facing a problem when importing DBCs, hwere the minumum and Maximum values of a custom attributes are set to 0. I'm working with the Vector CANDB++ Editor. The Editor ignores the validation of min max values for custom properties, when both are set to 0, and accept all Default values.

[image]

When importing such DBCs the DBCParser library returns 0 in such cases. Would it be possible to also ignore min/max validation by a setting or default behavior?

Best regards and you do a fantastic job,
 Julius

Whitehouse112, 2024-07-24:
Hi @JuRichter, 
you are definitely right: ignoring the validation should be the default behavior when both min and max values are set to zero. We will work on that for the next release.

Thanks
Regards

EFeru, 2024-07-26:
Fixed #66