EFeru/DbcParser #69: Update Target Frameworks

Uight, 2024-07-30:
Currently in the target frameworks of the project the following are listed:
netstandard2.0;net461;net5.0;net6.0

i would consider changing it a bit for the following reasons:
net461 is out of support since beginning of 2022 therefor atleast use net462 which is still officialy supported.
net5.0 is out of support;
net6.0 runs out of support this year but is okay for now;
netstandard2.0 is kind of tricky but as its currently supported you would probably keep supporting it.

net framework support:
https://learn.microsoft.com/de-de/lifecycle/products/microsoft-net-framework
.net support:
https://dotnet.microsoft.com/en-us/platform/support/policy
.net standard infos:
https://learn.microsoft.com/en-us/dotnet/standard
https://learn.microsoft.com/en-us/dotnet/standard/net-standard?tabs=net-standard-1-0#net-5-and-net-standard

I would like to change supported frameworks to:
netstandard2.0;net462;net6.0;net8.0

Adhara3, 2024-08-26:
Sound reasoable

Uight, 2024-08-28:
was solved with the merge of the pull request

closed 2024-08-28 by Uight