EFeru/DbcParser #70: Distinguish attributes (CustomProperties) between Network and Node attributes

ChrisFromNI, 2024-07-31:
Hi there,
there a dbc files which define global attributes or network attributes, e.g.
```
...
BA_DEF_  "Baudrate" INT 0 1000000;
...
BA_DEF_DEF_  "Baudrate" 500000;
...
```
When parsed, this attribute is added to all nodes' CustomProperties in Dbc.Nodes. However, each node may have unique Node attributes (defined with `BA_DEF_ BU_`) themselves. Now each node has CustomProperties which may contain Network and Node attributes. If I didn't miss any property/method, there is no way to distinguish between those two.

One way to accomplish would be a CustomProperties property on Dbc root level.

Uight, 2024-07-31:
@ChrisFromNI just to be sure. You have something like this:

// Attribute Definitions for Nodes
BA_DEF_ BU_ "NodeSoftwareVersion" STRING;
BA_DEF_DEF_ "NodeSoftwareVersion" "v1.0.0";

// Example usage for specific nodes
BU_: Node1 Node2 Node3

BA_ "NodeSoftwareVersion" BU_ Node1 "v2.1.3";
BA_ "NodeSoftwareVersion" BU_ Node2 "v2.0.5";

How i understand you is that you can see the Value "v2.1.3" when accessing the data for Node1 but you want to see that this is a Node specific value instead of a global value?

ChrisFromNI, 2024-07-31:
@Uight In addition to what you wrote, you can also have an attribute definition for the network. From my understanding the dbc file describes a hierarchy like Network>Nodes>Messages>Signals.

Network attributes look like that in my dbc file (not sure though if that's inside the CAN DBC specifications).

// Attribute Definitions for Network
BA_DEF_  "DBName" STRING ;
BA_DEF_  "Baudrate" INT 0 1000000;

BA_DEF_DEF_  "DBName" "DeviceXDatabase";
BA_DEF_DEF_  "Baudrate" 500000;

Those attributes get added to the CustomProperties dictionaries of all nodes. But there is no CustomProperties for the network level (which would equal to the root of DbcParserLib.Dbc). If you add your example lines to my lines and parse that you will get two Nodes with CustomProperties.

- Node1.CustomProperties will be:
   - <"NodeSoftwareVersion", CustomProperty with *.Value = "v2.1.3">
   - <"DBName", CustomProperty with *.Value = "">
   - <"Baudrate", CustomProperty with *.Value = "500000">
- Node2.CustomProperties will be:
   - <"NodeSoftwareVersion", CustomProperty with *.Value = "v2.0.5">
   - <"DBName", CustomProperty with *.Value = "">
   - <"Baudrate", CustomProperty with *.Value = "500000">

My point is that you cannot distinguish the network attributes from the node attributes. For my application I need to get the network attributes only.

Uight, 2024-08-23:
@ChrisFromNI would you expext these network properties to be in a seperate list of customProperties (named network properties) or would it be enough to see if a value of a property was specifically written and is not the default?

Adhara3, 2024-08-26:
Hi,

I think that if no specific BA_ or BU_ is indicated in the DEF_DEF_ properties should be added in a different root based container.
I would leave "network" out  here, it's too specific.

Cheers
A

ChrisFromNI, 2024-08-27:
@Uight I agree with @Adhara3's answer on your question:

> I think that if no specific BA_ or BU_ is indicated in the DEF_DEF_ properties should be added in a different root based container.
I would leave "network" out here, it's too specific.

Adhara3, 2024-08-27:
> @Uight I agree with @Adhara3's answer on your question:
> 
> > I think that if no specific BA_ or BU_ is indicated in the DEF_DEF_ properties should be added in a different root based container.
> > I would leave "network" out here, it's too specific.

Ok, I then renamed the issue to make it more clear