EFeru/DbcParser #96: Please support CANFD message parsing

pelva, 2024-11-07:
Below is the CANFD dbc generated by CANdb++, with the canfd flag added to the last line
It seems like this ： VFrameFormat={'StandardCAN':0,'ExtendedCAN':1,'StandardCAN_FD':14,'ExtendedCAN_FD':15}

```
VERSION ""

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_
	BO_TX_BU_
	BA_DEF_REL_
	BA_REL_
	BA_DEF_DEF_REL_
	BU_SG_REL_
	BU_EV_REL_
	BU_BO_REL_
	SG_MUL_VAL_

BS_:

BU_:

BO_ 2147483904 New_Message_1: 8 Vector__XXX
 SG_ New_Signal_2 : 8|8@1- (1,0) [0|0] "" Vector__XXX
 SG_ New_Signal_1 : 0|8@1- (1,0) [0|0] "" Vector__XXX

BA_DEF_  "MultiplexExtEnabled" ENUM  "No","Yes";
BA_DEF_ BO_  "CANFD_BRS" ENUM  "0","1";
BA_DEF_  "DBName" STRING ;
BA_DEF_  "BusType" STRING ;
BA_DEF_ BU_  "NodeLayerModules" STRING ;
BA_DEF_ BU_  "ECU" STRING ;
BA_DEF_ BU_  "CANoeJitterMax" INT 0 0;
BA_DEF_ BU_  "CANoeJitterMin" INT 0 0;
BA_DEF_ BU_  "CANoeDrift" INT 0 0;
BA_DEF_ BU_  "CANoeStartDelay" INT 0 0;
BA_DEF_ BO_  "VFrameFormat" ENUM  "StandardCAN","ExtendedCAN","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","StandardCAN_FD","ExtendedCAN_FD";
BA_DEF_DEF_  "MultiplexExtEnabled" "No";
BA_DEF_DEF_  "CANFD_BRS" "1";
BA_DEF_DEF_  "DBName" "";
BA_DEF_DEF_  "BusType" "";
BA_DEF_DEF_  "NodeLayerModules" "";
BA_DEF_DEF_  "ECU" "";
BA_DEF_DEF_  "CANoeJitterMax" 0;
BA_DEF_DEF_  "CANoeJitterMin" 0;
BA_DEF_DEF_  "CANoeDrift" 0;
BA_DEF_DEF_  "CANoeStartDelay" 0;
BA_DEF_DEF_  "VFrameFormat" "StandardCAN";
BA_ "BusType" "CAN FD";
BA_ "DBName" "canfd";
BA_ "VFrameFormat" BO_ 2147483904 15;

```

Adhara3, 2024-11-09:
Hi,

`VFrameFormat` is not standard stuff, I mean this is quite Vector specific. I am a bit reluctant to add specific properties in the object based on non standard custom properties. 
What if the  `VFrameFormat` definition (`BA_DEF_ BO_`) and /or value (`BA_DEF_DEF_` and `BA_ `)  is missing? What is default?
Moreover, the list is quite open
```
ENUM  "StandardCAN","ExtendedCAN","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","reserved","StandardCAN_FD","ExtendedCAN_FD"
```
What about all those `reserved`?

We could eventually add an extension method for convenience, something like
```cs
public static string VFrameFormat(this Message message)
{
  // Checks for the VFrameFormat property and returns null if no property is found of the value if found
}
```
I would prefer keeping the `string` but let's say we go go for an `enum` then the values should be
```cs
public enum VFrameFormat
{
  NotSet = 0,
  Unknown, // Property set but value not recognized
  StandardCAN,
  ExtendedCAN,
  StandardCAN_FD,
  ExtendedCAN_FD
}
``` 

A

pelva, 2024-11-11:
Adding extension methods is fine.

Adhara3, 2024-11-11:
Hi,

the good thing about extension method is that you can write it yourself in your code _extending_ current functionality, at least until we relese the new version.

```cs
public static VFrameFormat FrameFormat(this Message message)
{
    if (message.CustomProperties.TryGetValue("VFrameFormat", out var frameFormat) == false)
        return VFrameFormat.NotSet;

    switch (frameFormat.StringCustomProperty.Value)
    {
        case "StandardCAN":
            return VFrameFormat.StandardCan;
        case "ExtendedCAN" :
            return VFrameFormat.ExtendedCan;
        case "StandardCAN_FD" : 
            return VFrameFormat.StandardCanFd;
        case "ExtendedCAN_FD":
            return VFrameFormat.ExtendedCanFd;
        default:
            return VFrameFormat.Unknown;
    }
}

public enum VFrameFormat
{
    NotSet = 0,
    Unknown, // Property set but value not recognized
    StandardCan,
    ExtendedCan,
    StandardCanFd,
    ExtendedCanFd
}

```

Cheers
A

pelva, 2024-11-12:
That's awesome, thanks!