EFeru/DbcParser #108: [Feature Request] Output to DBC String

c0nstexpr, 2025-07-10:
Just like following example:
```C#
Dbc dbc = new();
// ...
var dbcStr = dbc.ToString();
File.WriteAllText("test.dbc", dbcStr);
```

Adhara3, 2025-07-10:
Hi,

This is a huge one and a bit out of the scope of a parser which is basically just a reader.

My 2 cents
A

pelva, 2025-09-17:
The workload for generating DBC files is indeed substantial. By the way can we update the NuGet package? The latest version on NuGet was released on September 7, 2024, but there have been some useful updates recently.

Adhara3, 2026-01-09:
> The workload for generating DBC files is indeed substantial

I'm well aware.

> By the way can we update the NuGet package? The latest version on NuGet was released on September 7, 2024, but there have been some useful updates recently.

Done